UnchartedBull/OctoDash #1451: Installing Octodash changes video output from TFT touch screen back to HDMI

jakebullet70, 2021-02-20:
**What doesn't work?**
I have a working OctoPi install with a Aliexpress 3.5in TFT using LCD-show drivers. I have loaded the drivers and all is good. My LCD is working fine. After I load OctoDash with your script it installs fine except the output changes from my little TFT screen back to the HDMI port. There are no further updates to the little TFT screen. HDMI is working fine and I am prompted to setup OctoDash from there.

**What did you already try?**
Uninstalled OctoDash. Little TFT screen starts working again and I can interact with the CL. installed OctoDash again and it does the same thing, back to HDMI.

**General Information:**
 - Raspberry PI4 - AliExpress 3.5 inch TFT with goodTFT drivers
 - Fresh OctoPi 0.18
 - OctoDash pulled today
 - OctoPrint Version 1.53
 - TFT drivers. https://github.com/goodtft/LCD-show

**Additional context**
When booting you can see the boot start on HDMI, switch to the TFT screen and then back to HDMI when Octodash starts.

carl1961, 2021-02-20:
maybe this link will help, you need to get to terminal console screen .
https://digitalab.org/2019/04/how-to-setup-3-5-inch-touch-screen-on-raspberry-pi/

jakebullet70, 2021-02-20:
Thanks. I have seen this. The 3.5in TFT works fine. Its only after the Octodash install that I have any issues.

carl1961, 2021-02-20:
I was thinking if you reinstalled the TFT driver after octodash , it should revert to your TFT. I could not read anything in the install script that installed a display

jakebullet70, 2021-02-21:
Oh I see, will try that when I get home.

jakebullet70, 2021-02-21:
Reinstalled the TFT drivers **after** Octodash was installed and rebooted, got an FBTurbo error when Octodash started and it failed but it stayed on the on the small TFT screen. 
After some searching I found this. https://www.raspberrypi.org/forums/viewtopic.php?t=162323

I added this to my '/etc/X11/xorg.conf.d/99-calibration.conf'

Section "Device"
# Cheapo 3.5", framebuffer 1 (goodTFT)
Identifier "uga"
driver "fbdev"
Option "fbdev" "/dev/fb1"
Option "ShadowFB" "off"
EndSection

Section "Monitor"
# Primary monitor. Cheapo (goodTFT) 480x320
Identifier "WSSP"
EndSection

Section "Screen"
Identifier "primary"
Device "uga"
Monitor "WSSP"
EndSection

Section "ServerLayout"
Identifier "default"
Screen 0 "primary" 0 0
EndSection

Thanks for the help, All is good now. You can close the ticket.

closed 2021-02-21 by jakebullet70