UnchartedBull/OctoDash #1458: Filament Change with M600 / unpause via screen / i3 Mega-S DGUS-Display

hawkeyexp, 2021-02-22:
Hi,

i'm using an anycubic i3 mega-s with the new blue yellow display running with marlin firmware.
Filament change is well triggered and unloaded works fine also load but:

after unload printer is waiting for user input to resume but no dialog etc. on printer display.
it would be great if you can add a button (possible optional to be enabled via plugin settings) which allows to send the g-code M108 to the printer which breaks the wait queue and starts loading.

Manually entering M108 via terminal works perfect and breaks the wait loop and starts loading the filament.

UnchartedBull, 2021-03-02:
Is this occurring during printing or during the filament change process?

hawkeyexp, 2021-03-02:
it is needed during filament change - after unload system is waiting for user input without any display message on printer screen - i can se the wait loop on terminal output and M108 breaks the loop and filament load starts.

thegarbz, 2021-03-03:
This will happen on all printers with Marlin firmware, can confirm on the Prusa's as well. Purely as a matter of interest, what is the benefit here? When you unload the filament don't you need to be at the printer to remove it anyway? Can't you just acknowledge it at the time or do you have some automated loader/unloader?

hawkeyexp, 2021-03-03:
The benefit is easy to explain: i can fully use this function.

the i3 mega series has no push button and M600 is waiting for user input after filamente unload to start the loading oft the new inserted filament - M108 brakes the wait loop like the button press on printers where it exists - so in this case M108 replaces the button press and makes the Filamente change fully working - i created a custom button as a temporary solution and it works like a charm.

UnchartedBull, 2021-03-10:
Mhh, why aren't you using the integrated procedure then? That would work without any input from the printer.

hawkeyexp, 2021-03-11:
It has two reasons:

unload speed is really slow compared to buildin with M600 - it seems he is always using the slow speeds in my case.
M600 is a general part of marlin and after unload printer is signaling with beep sound that he has unloaded the filament and is waiting for the new one.
It would be nice if you can extend the settings with an additional setting while M600 is used which enables a dialog which shows that printer is waiting for user with an ack button triggering the M108 command which let's progress the printer with his loading.

UnchartedBull, 2021-03-11:
Have you tried increasing the speed in the settings then? Also OctoDash will show you the current speed right under the progress bar. You might also have the speed setup incorrectly in OctoDash see here: https://github.com/UnchartedBull/OctoDash/wiki/Troubleshooting#filament-change-is-slower-than-specified-in-configuration.

hawkeyexp, 2021-03-12:
I tried it out but system is still loading and unloading in slow mode.
i prefer the buildin M600 from firmware which works well.

Poeschl, 2021-03-23:
I can confirm that the Unloading in Octodash is indeed slow. During unloading and loading for a few seconds my `feedSpeed` value is shown and for like 80% of the filling progressbar the `feedSpeedSlow` gets displayed.

Filament is still unloading while octodash is already waiting for the "to-be-loaded" Filament.

UnchartedBull, 2021-03-23:
Could you please double check the settings in your OctoPrint printer profile and post them here alongside the speeds you have configured in OctoDash? Tahnks?

This tab:
[image]

Poeschl, 2021-03-23:
That's it. I just checked them and still had 300 mm/min there (5mm/s). Increased the feedrate and its working fine. 👍

hawkeyexp, 2021-04-05:
works

closed 2021-04-05 by hawkeyexp