UnchartedBull/OctoDash #1472: bug that crashes the whole thing

StrawCor, 2021-02-23:
Hi Guys,
Love the software and new to this. I have the talent to crash things unintentionally. I crashed the octopi and octodash by inputting a number.
to Reproduce**
ok, to reproduce the bug ( I have only tried this once), all I did was input 3.33 in the Z axis as the wished speed. 

**Expected behavior**
What happened was a screen appeared saying " Your config is invalid" it then names the printer zSpeed as the problem.
At this point the touch screen no longer works, in fact, feck all works. It will not show up in the browser. If I remove the card and put it in my Linux box, I cannot even open the fsroot file. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General Information:**
 - Raspberry PI 2
 - Octopi 18
 - newOctodash
 - Waveshare 3.5 inch hdmi screen.

 I have a Logitech hd webcam plugged in, logitech Keyboard and Ethernet cable.

**Additional context**
I do not think that the Logitech accessories are causing the problem. It is most probably just me, a Feck tard of 50 years old. Again, nice job with Octodash.

UnchartedBull, 2021-03-02:
Damn, I think you managed to enter the only invalid input in the text field that isn't checked. OctoDash defines zSpeed as an integer (only whole numbers). You can either remove your config and start from scratch (for OctoDash only) or edit your current config and change zSpeed to a whole number (i.e. 3). https://github.com/UnchartedBull/OctoDash/wiki/Setup-&-Settings#manually-adjusting-the-settings

StrawCor, 2021-03-03:
Hi Timon,
Thanks for the reply. I did a new install as the sd card I was using seemed too slow and was hacking the picture from the webcam. I was able to load all the programs I wanted at the start, and now with the V10 sd Card it works better than it did before. Again great job on the software.

closed 2021-03-03 by StrawCor