UnchartedBull/OctoDash #1497: OctoDash to OctoPi connection issues persist

M1dn1ghtN1nj4, 2021-02-28:
**What doesn't work?**
As of today, using the most recent official release of OctoPi, OctoDash is still having issues connecting to OctoPrint.

**What did you already try?**
I started from a virgin install of OctoPi 0.18.0, and installed the both the latest build of my OctoDash builds, and the last official release from Nov 2020 (2.1.2).

**General Information:**
 - Pi 4, 1GB
 - OctoPi 0.18.0 (official build, not a nightly)
 - OctoDash 2.1.2 (Nov 2020)
 - OctoPrint 1.5.3

**Additional context**
This happened before, but now I can't seem to get past it.  The OctoDash setup switches between saying auto-login not supported, create API key and paste below, and connection error, cannot connect to OctoPrint instance on this machine.

I've also tried with my current build of OctoDash (yesterday) and checking/unchecking the websockets option.  That has had zero effect so far.  I'm going to try my setup one more time, but as of now, I can't find a combination of OctoPi, OctoPrint, and OctoDash that successfully work together.  Until I can get this sorted out, I'll have to run my machine headless.

UnchartedBull, 2021-03-02:
Could you check if you can use the API fine with Postman or cURL? Also does OctoDash work once setup has been completed?

M1dn1ghtN1nj4, 2021-03-03:
Well, that's the thing, I can't get into OctoDash because I can't finish the setup.  It also doesn't seem to want to accept any API key I give it.  I also honestly know zero about Postman or cURL, so I'd need some direction on those.

Ill try a bit more over the next couple days.  I have a lot to do right now, so this gets tested when I have a free minute.  It could be that I'm just a fluke, and my setup is being stupid.  Don't take this as everybody is getting this problem.  I'll get more info to you as soon as I can.

UnchartedBull, 2021-03-05:
Some people have reported similar issues to this, so it seems like this is a bigger problem.

Postman is basically a little program that allows you to send API requests via a nice UI. The UI should be pretty self explanatory, you can get info about the OctoPrint API here: https://docs.octoprint.org/en/master/api/general.html. I'm basically trying to figure out whether this is related to a bug with the OctoPrint API or with OctoDash.

antonyburden, 2021-03-14:
Hello, I can also confirm, that I upgraded from official build nov 2020 to midnightninjas latest nightly build. It just will not connect, tried removing config file but won’t connect. 

Also removed completely and tried reinstalling the official build, but same issue. 

Latest octopi and octoprint as above.

UnchartedBull, 2021-03-18:
I think this is most likely due to the mdns name not being setup correctly. I have a fix planned for v3 which checks whether you're trying to connect to the local octoprint, this will then ask you, whether you want to use `localhost` instead of `octopi.local`. Has anyone tried connecting via `localhost` here yet and can confirm that this works fine?

antonyburden, 2021-03-18:
I have tried, 127.0.0.1, localhost, octopi.local ports 80 and 5000

antonyburden, 2021-03-25:
Hi, any news on this?

UnchartedBull, 2021-03-26:
Please test the beta, that is available on the Discord. The beta hopefully fixes this issue. If anyone can confirm this, this can be closed here I think.

M1dn1ghtN1nj4, 2021-03-26:
I'm not at a point where I can test right now.  I have 2 very large orders
that I need to fill.  Can't have my machine down until they are finished.
Anyone else able to test?

antonyburden, 2021-03-26:
Same as before, uninstalled, removed config file, installed latest beta.
Setup started, it finds the OctoPrint instance but will not get the API key, have also manually entered 127.0.0.1 and 192.168.1.38 for both port 80 and 5000. I also entered the API manually. Same error at end of setup "0 unknown"

antonyburden, 2021-03-26:
Working now I've enabled CORs in OctoPrint API settings :-)

UnchartedBull, 2021-03-29:
Good to know. Closing this here since it seems to be fixed. I've also added a section to the troubleshooting guide: https://github.com/UnchartedBull/OctoDash/wiki/Troubleshooting#octodash-cant-connect-to-octoprint-0-unknown. @M1dn1ghtN1nj4 if you're jobs are done and you have some time to experiment feel free to try it again and reopen this issue if it still doesn't work for you.

closed 2021-03-29 by UnchartedBull