UnchartedBull/OctoDash #1523: error: cannot retrieve job status when printing from SD

Photogad, 2021-03-05:
**What doesn't work?**
When I print from SD, I get the error on my screen. It doesn't go away, keeps coming back even if I tap on it, and even if the print job is done. The only thing that makes it go away is restarting the Pi. I've only had OctoDash for 4 days now, and for the first 2 days, I didn't have this error at all. It just randomly started happening, no changes were made to the Pi.

**What did you already try?**
Nothing yet, idk where to begin or how to find any logs (I'm a Pi n00b).

**General Information:**
 - Hardware Pi 3B+ with a proper power plug.
 - OctoPi/OctoPrint latest version
 - OctoDash Version is latest I'm assuming because I've only had it for 4 days.

**Additional context**

[image]

UnchartedBull, 2021-03-10:
duplicate of #941. Will be fixed with v3, there currently seem to be a couple issues regarding sd cards. Nothing you can do here except wait unfortunately.

closed 2021-03-10 by UnchartedBull

Photogad, 2021-03-10:
> duplicate of #941. Will be fixed with v3, there currently seem to be a couple issues regarding sd cards. Nothing you can do here except wait unfortunately.

Well, it's not a duplicate of that issue entirely. Mine is doing it even when the .gcode file is on the root of the SD card. In fact, I have no subfolders on my SD card at all presently. And, Octodash was working fine for 2 days without this problem; clearly something changed. I did add a second camera to my Pi installation, maybe that is causing some kind of conflict.

UnchartedBull, 2021-03-11:
Might not be 100% the same but will get fixed with the same patch. OctoDash will then rely on OctoPrint providing the data, rather than OctoDash requesting it. So this error can't appear anymore :)

Photogad, 2021-03-12:
> Might not be 100% the same but will get fixed with the same patch. OctoDash will then rely on OctoPrint providing the data, rather than OctoDash requesting it. So this error can't appear anymore :)

Ok perfect mate. Will patiently await for the update.