UnchartedBull/OctoDash #1526: The automatic installation doesn`t work..

lijiajian1994, 2021-03-06:
RaspberryPi 4B+Octopi v1.8 system.
Execute the automatic installation script,and after 2 seconds,it pops up directly.
May I ask why?
[image]

lijiajian1994, 2021-03-10:
> 您的控制台屏幕截图未显示任何有用的信息，请发布正在发生的错误。

The first command is an error, after entering the automatic installation instruction, there is no error, can not continue to install, but directly jump back to the state of the input instruction.

UnchartedBull, 2021-03-11:
please follow the manual installation instructions then: https://github.com/UnchartedBull/OctoDash/wiki/Installation#manual-installation. Could you please post the output of `echo $TERM` here as well?

closed 2021-04-18 by stale[bot]