UnchartedBull/OctoDash #1527: I cannot make Octodash work on RPi

gheorghitacristian, 2021-03-06:
**What doesn't work?**
I cannot boot into Octodash after installing it. After installing Octodash on a newly installed Octopi i get the error in the attached picture.

**What did you already try?**
I only installed Octodash using the command in the instructions, I selected the plugins, I selected the option to reboot after it finished, and after reboot I get the screen as per attached attached picture.

**General Information:**
 - Hardware : Raspberry Pi 3B+
 - OctoPrint 1.5.2 // Python 3.7.3 // OctoPi 0.18.0
 - OctoDash Version - latest version on Github

[image]

JeremyProffitt, 2021-03-09:
Same issue, Raspberry Pi 4 - 4GB, fresh copy of octoprint, ran apt -y update; apt -y upgrade-full then the installer.

gheorghitacristian, 2021-03-09:
> Same issue, Raspberry Pi 4 - 4GB, fresh copy of octoprint, ran apt -y update; apt -y upgrade-full then the installer.

What type of screen do you have? Do you use GPIO or HDMI port to connect the screen to the board?

JeremyProffitt, 2021-03-09:
HDMI screen, 7" capable of scaling.  So, a standard HDMI monitor.

LeParan0, 2021-03-09:
I try this : 
sudo apt-get install git build-essential xorg-dev xutils-dev x11proto-dri2-dev
sudo reboot
sudo apt-get install libltdl-dev libtool automake libdrm-dev
sudo reboot
git clone https://github.com/ssvb/xf86-video-fbturbo.git
cd xf86-video-fbturbo
autoreconf -vi
./configure --prefix=/usr
make
sudo make install
sudo cp xorg.conf /etc/X11/xorg.conf
sudo reboot

but dont work better, then i try this : 
git clone https://github.com/waveshare/LCD-show.git (or this: git clone https://github.com/goodtft/LCD-show.git)
chmod -R 755 LCD-show
cd LCD-show/
chmod +x LCD35-show
sudo ./LCD35-show

And its working ! 
But i dont know if ONLY the "chmod +x LCD35-show" part work

Aprextext, 2021-03-17:
I was having the same issue, I've tried LeParan0's fix, and that has got me from the same screen as in the OP image, to a VNC screen on the display, although there was a brief splash screen. If I try and run octodash from an SSH window, it says "Segmentation Fault". I am using a pi4 running buster (precompiled straight from octoprint) and using an elegoo 3.5inch touchscreen via SPI.

JeremyProffitt, 2021-03-17:
Same problem here, fix didn't work, this sounds like a code issue with a
new version of something.

Aprextext, 2021-03-17:
I suddenly thought "What if VNC is merely blocking the screen", so I went in via SSH, turned it off, and lo and behold, it worked. And it's beautiful. The file manager is easier to use than the web UI. Thankyou UnchartedBull

UnchartedBull, 2021-03-18:
Thanks @Aprextext 

@JeremyProffitt are you still having issues? The release file wasn't updated, so there were no code changes in the latest release. Only thing I can think of is something miss-configured or some dependency got updated and doesn't play nice with the rest anymore. It's the former in 99% of the cases though.

gheorghitacristian, 2021-03-19:
Its a week now keep on trying. I cannot waste more time on it anymore. Thanks for pitching in for support.

closed 2021-03-19 by gheorghitacristian