UnchartedBull/OctoDash #1528: Octodash screen is completely white

Michael1331-crypto, 2021-03-06:
Hello,
Today I set up my OctoDash. It worked completely fine for about 20 times. But then I turned it off and on again and now the screen stays completely white! 

I hope I don't have to disassemble everything again to flash it new :/

I use a Raspberry Pi 3B
elegoo 3,5" Touch display
newest OctoPrint version
newest OctoDash version

I was so happy it is working and now this ... I hope somebody can help me :)
thanks in advance :)

UnchartedBull, 2021-03-10:
Is the console showing up during the boot or is the screen always white?

Michael1331-crypto, 2021-03-19:
It is always white!

UnchartedBull, 2021-03-19:
Then it's most likely a hardware issue. Please double check your wiring and try a different screen.

closed 2021-06-02 by stale[bot]