UnchartedBull/OctoDash #1531: Show Infoscreen while system is autoleveling (G29)

hawkeyexp, 2021-03-08:
It would be nice if system is showing an info screen while autoleveling is running like a print progress dialog and showing the colelcted matrix or possible a graphical map) after g29 is done.

hawkeyexp, 2021-03-11:
H, not fully - a bit stripped down - notify while process is running is the main request.
Showing the results and a map is the topping if possible :-)

UnchartedBull, 2021-03-11:
Mhh ok, maybe I'm missing something here, but can't you just look at the printer to check if it's leveling? I'm a little hesitant here since this would be fairly hard to implement. OctoPrint doesn't provide this information natively anywhere and OctoDash is completely unaware of the GCode that is currently being printed. So this would involve another plugin and a lot of work. If there is any easy way to do this with OctoPrint I'm more than happy to go forward with this.

hawkeyexp, 2021-03-11:
ah ok - that makes it harder if can't monitor the g-code to have a look for the g29 command and it's reply of result to set start and end trigger. I will dive deeper in if i have an idea or finding an asy way to adapt.

UnchartedBull, 2021-03-11:
The websocket integration is great news! So this will definitely be possible, after release of v3 though. And I still need to work out on how this actually displayed. If you have any ideas let me know :)

hawkeyexp, 2021-03-12:
perfect - and fast response from jneilliii :-) So with have to ait a bit :-9