UnchartedBull/OctoDash #1547: Integration with OctoPrint-TimeToFilament plugin

sbussinger, 2021-03-11:
I'd love to see an integration with OctoDash and the OctoPrint-TimeToFilament plugin. This is a great plugin that displays the time until the next filament change (or perhaps a pause to insert a nut) in OctoPrint.

The plugin is at https://plugins.octoprint.org/plugins/TimeToFilament/ and the OctoPrint output looks like:
[image]

I'd love to see that Filament Change countdown displayed in OctoDash while a print job is running.

UnchartedBull, 2021-03-11:
This plugin doesn't have an API, so it won't be possible to implement unfortunately. But #1317 might be the same, just with a different plugin(changeFilamentTimeLeft).

closed 2021-03-11 by UnchartedBull

sbussinger, 2021-03-11:
Oh, I had no idea that DisplayLayerProgress even had that feature. It's not much documented on it's pages. I'll dig into that and see if it does what I want (it appears to). In that case this is a duplicate of #1317 

I'd like it to trigger on @pause as well as M600, but I'll talk to them about that. It shouldn't affect your integration with it.