UnchartedBull/OctoDash #1549: [FR] Switch from M600 to M701 and M702

lightmaster, 2021-03-12:
**Is your feature request related to a problem? Please describe.**
When doing a filament change with no filament currently loaded, you must wait for it to fully unload an invisible roll before being able to load the new roll.

Conversely, when wanting to unload the filament to store and are not planning to load a new roll, you must wait for it to fully load the new invisible roll. (hitting skip button does not stop the loading)

**Describe the solution you'd like**
Switch from using `M600` to using `M701` and `M702` so you can pick if you want to just load or unload the filament without having to do both back to back. Also when using the OctoDash unload and load, have options to do just one or the other instead of forcing both.

**Link other projects**
If you want OctoDash to include other OctoPrint plugins or similar please link them here.

**Additional context**
Add any other context or screenshots about the feature request here.

thegarbz, 2021-03-16:
So there's no way to automate this since AFAIK Octodash doesn't keep the state of the filament handy. However personally I've assigned M702 to the Custom Actions in Octodash precisely so I can unload the printer at the end of the print.  https://github.com/UnchartedBull/OctoDash/wiki/Custom-Actions
I use the Angle Double Up icon for this button.  https://fontawesome.com/icons/angle-double-up?style=solid

The only way I can see this being easily implementable is if the Filament management screen gives you the choice between loading, unloading or changing filament.

lightmaster, 2021-03-16:
> The only way I can see this being easily implementable is if the Filament management screen gives you the choice between loading, unloading or changing filament.

Exactly what I was thinking, essentially to replicate the screen from Marlin. You can pick
1. `Load Filament` which sends the `M701` command and then has a prompt to send a `M108` to stop the purge early.
2. `Unload Filament` to send `M702` command.
3. `Change Filament` to send `M600`

For load and change filament, it changes the current filament in Filament Manager/SpoolManager.

UnchartedBull, 2021-03-18:
Not 100% sure about this one. The main idea behind OctoDash is to be simple and very easy to use. I think having even more buttons on the filament screen won't support this idea. I also think it wouldn't save too much time, you would still need for the warmup and all that, so it's just the unloading which should only be a couple of seconds.

With the next version you have the shortcut filament selection (hold 3s on a filament to skip the internal change process and just notify FilamentManager of the new filament), so in M701 / M702 case you could use the internal display and for "normal" changes you can use the full-fledged change process in OctoDash.

closed 2021-03-18 by UnchartedBull

lightmaster, 2021-03-18:
> With the next version you have the shortcut filament selection (hold 3s on a filament to skip the internal change process and just notify FilamentManager of the new filament), so in M701 / M702 case [...]

The ability to change filaments in filament manager without needing unload/load is sufficient for my purposes. Load (without unload) from printer LCD and switch filaments from OctoDash.