UnchartedBull/OctoDash #1552: M0 Pause not executing in custom commands

FlyingT, 2021-03-13:
I wanted to use Custom Actions with GCode for manual bed leveling, but while the move commands are being executed, the M0 pause isnt. 

The GCode I used can be seen here: [https://pastebin.com/TdKZrNLw](https://pastebin.com/TdKZrNLw)

Removed all comments to make it into one long line and this is what I ended up with:

```
"icon": "ruler-vertical",
"command": "G90; G28; G1 Z5; G1 X32 Y36 F3000; G1 Z0; M0; G1 Z10; G1 X32 Y206 F3000; M0; G1 Z5; G1 X202 Y206 F3000; G1 Z0; M0; G1 Z5; G1 X202 Y36 F3000; G1 Z0; M0; G1 Z5; G1 X117 Y121 F3000; G1 Z0; M0; G1 Z5; G1 X32 Y206 F3000; G1 Z0; M0; G1 Z5; G1 X202 Y206 F3000; G1 Z0; M0; G1 Z5; G1 X202 Y36 F3000; G1 Z0; M0; G1 Z5; G1 X32 Y36 F3000; G1 Z0; M0; G28; M84",  
"color": "#44bd32",  
"confirm": false,  
"exit": true,
```  
Did I do something wrong os is this an bug in OctoDash?

FlyingT, 2021-03-14:
M25 Pause also doesnt seem to work

UnchartedBull, 2021-03-18:
I think is most likely due to OctoPrint. Have you tried the steps mentioned here: https://github.com/OctoPrint/OctoPrint/issues/1513?

FlyingT, 2021-03-18:
Seems to be outdated, but found the solution:

OctoPrint Settings > Serial Connection > Firmware & Protocol > Protocol fine tuning > Blocked commands
Then remove M0 (and M1 if you feel like it) from the list

It works now, thanks!

closed 2021-03-18 by FlyingT