UnchartedBull/OctoDash #1560: Smaller steps (0.05 & 0.1)

seppelicous, 2021-03-16:
Hi, 
right now the axis movement is limited to fixed values (1, 5, 10, 50... mm).
Is it possible to edit this values, for example when I level the bed, I like to lift the Z axis in small steps.
Or is this already possible?

best regards

UnchartedBull, 2021-03-18:
It's possible while printing. For manual bed leveling there is issue #312, for automatic #1092. So duplicate of one of those.

seppelicous, 2021-03-23:
I assume you do not understand what I mean. I do not want do modify the Z height during printing. I just want to set the height in smaller steps, like 0.025, 0.05 and 0.1mm for leveling the bed.
As I see so far, Octodash only allows 1mm and higher height steps?