UnchartedBull/OctoDash #1582: OctoDash does not boot

krisiks-jpg, 2021-03-20:
Hi, I have gone through many guides and similar threads, my problems is that OctoDash does not boot at all.

Xorg.0.log

> [    32.806] 
X.Org X Server 1.20.4
X Protocol Version 11, Revision 0
[    32.809] Build Operating System: Linux 5.4.0-54-generic armv8l Raspbian
[    32.810] Current Operating System: Linux octopi 5.10.17-v7+ #1403 SMP Mon Feb 22 11:29:51 GMT 2021 armv7l
[    32.810] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=640 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=tty1 console=ttyAMA0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait fbcon=map:10 fbcon=font:ProFont6x11 logo.nologo
[    32.816] Build Date: 15 January 2021  02:03:57PM
[    32.817] xorg-server 2:1.20.4-1+rpt2+deb10u3 (https://www.debian.org/support) 
[    32.818] Current version of pixman: 0.36.0
[    32.819] 	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
[    32.819] Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    32.827] (==) Log file: "/var/log/Xorg.0.log", Time: Sat Mar 20 17:23:46 2021
[    32.840] (==) Using config directory: "/etc/X11/xorg.conf.d"
[    32.842] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[    32.848] (==) No Layout section.  Using the first Screen section.
[    32.848] (==) No screen section available. Using defaults.
[    32.848] (**) |-->Screen "Default Screen Section" (0)
[    32.848] (**) |   |-->Monitor "<default monitor>"
[    32.854] (==) No device specified for screen "Default Screen Section".
	Using the first device section listed.
[    32.854] (**) |   |-->Device "Allwinner A10/A13 FBDEV"
[    32.854] (==) No monitor specified for screen "Default Screen Section".
	Using a default monitor configuration.
[    32.854] (==) Automatically adding devices
[    32.854] (==) Automatically enabling devices
[    32.854] (==) Automatically adding GPU devices
[    32.861] (==) Max clients allowed: 256, resource mask: 0x1fffff
[    32.863] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[    32.864] 	Entry deleted from font path.
[    32.864] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[    32.864] 	Entry deleted from font path.
[    32.864] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[    32.864] 	Entry deleted from font path.
[    32.864] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[    32.864] 	Entry deleted from font path.
[    32.864] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[    32.864] 	Entry deleted from font path.
[    32.864] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[    32.864] 	Entry deleted from font path.
[    32.864] (==) FontPath set to:
	/usr/share/fonts/X11/misc,
	built-ins
[    32.864] (==) ModulePath set to "/usr/lib/xorg/modules"
[    32.864] (II) The server relies on udev to provide the list of input devices.
	If no devices become available, reconfigure udev or disable AutoAddDevices.
[    32.864] (II) Loader magic: 0x1fdf80
[    32.864] (II) Module ABI versions:
[    32.864] 	X.Org ANSI C Emulation: 0.4
[    32.864] 	X.Org Video Driver: 24.0
[    32.864] 	X.Org XInput driver : 24.1
[    32.864] 	X.Org Server Extension : 10.0
[    32.866] (--) using VT number 2
[    32.866] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
[    32.867] (II) no primary bus or device found
[    32.867] (II) LoadModule: "glx"
[    32.869] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[    32.931] (II) Module glx: vendor="X.Org Foundation"
[    32.931] 	compiled for 1.20.4, module version = 1.0.0
[    32.931] 	ABI class: X.Org Server Extension, version 10.0
[    32.931] (II) LoadModule: "fbturbo"
[    32.932] (II) Loading /usr/lib/xorg/modules/drivers/fbturbo_drv.so
[    32.936] (II) Module fbturbo: vendor="X.Org Foundation"
[    32.936] 	compiled for 1.20.3, module version = 0.5.1
[... 133 lines cut ...]
> X.Org X Server 1.20.4
X Protocol Version 11, Revision 0
Build Operating System: Linux 5.4.0-54-generic armv8l Raspbian
Current Operating System: Linux octopi 5.10.17-v7+ #1403 SMP Mon Feb 22 11:29:51 GMT 2021 armv7l
Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=640 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=tty1 console=ttyAMA0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait fbcon=map:10 fbcon=font:ProFont6x11 logo.nologo
Build Date: 15 January 2021  02:03:57PM
xorg-server 2:1.20.4-1+rpt2+deb10u3 (https://www.debian.org/support)
Current version of pixman: 0.36.0
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.1.log", Time: Sat Mar 20 17:42:02 2021
(==) Using config directory: "/etc/X11/xorg.conf.d"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
modprobe: FATAL: Module g2d_23 not found in directory /lib/modules/5.10.17-v7+
[13540:0320/174202.526054:FATAL:electron_main_delegate.cc(254)] Running as root without --no-sandbox is not supported. See https://crbug.com/638180.
Trace/breakpoint trap
xinit: connection to X server lost
waiting for X server to shut down XIO:  fatal IO error 4 (Interrupted system call) on X server ":1"
after 157 requests (157 known processed) with 0 events remaining.
(II) Server terminated successfully (0). Closing log file.

Have done many other manipulations, install, reinstalls and updates, did not help at all. Can anyone say in which direction to go? Thank You!

krisiks-jpg, 2021-03-20:
Tried 
`export DISPLAY=:0`
And then 
`ratpoison`
Console says: 
> ratpoison: There can be only ONE.

LCD is 3.5inch GoodTFT display.

Will-wastelander, 2021-03-21:
Did you install the LCD drivers ?

https://github.com/goodtft/LCD-show

pciavald, 2021-03-21:
@krisiks-jpg  it seems you're somehow launching X twice. Can you please post all the commands you run to get there ?

JeremyProffitt, 2021-04-01:
This is identical to my issue.  https://github.com/UnchartedBull/OctoDash/issues/1632

my testing was done with octopi, there are not two copies of x running, review the above ticket, this is a chromium issue with a newer release.

closed 2021-06-05 by stale[bot]