UnchartedBull/OctoDash #1586: Configure window size and position

pciavald, 2021-03-21:
**Is your feature request related to a problem? Please describe.**

For our application, we need to let the top bar (lxpanel) appear above OctoDash. We do not want to hide a part of octodash and making lxpanel appear above octodash's layer is rather complex anyway.

**Describe the solution you'd like**

I've implemented the settings we need in main.js on our branch, but that's just one more hard-coded config with --dev, --big and the fullscreen setting. Functionally, I think the most elegant situation would be to read OctoDash config file before creating the window to check if a custom size and position of the window is defined. If that seems like the right way for you too, i can push a PR.

pciavald, 2021-03-23:
I see you've already added an helper to read the config, i'll make a proposition based on this !

UnchartedBull, 2021-03-28:
Not 100% sure about this for the main branch. This most likely will lead to even more problems reported about screens not working / the resolution being off / oversized images / ... . So this most likely will be something for your custom fork or is there another use case I'm missing here?

pciavald, 2021-03-28:
Actually there's a little refactor to main.js i've been thinking about, that makes resolution management leaner... so that may as well help with other potential issues, especially reproducing the issue of an user with specific resolution. I'll send a proposition and you can decide whether you want it in upstream or not :)