UnchartedBull/OctoDash #1592: octodash values not updating since websocket update

pciavald, 2021-03-23:
**Describe the bug**

hotend, bed and fan values not updating, print status not updating.

**To Reproduce**
Steps to reproduce the behavior:

- clone latest main (c26a901)
- package app
- install deb
- start and connect
- click hotend, set temperature
- temperature is set correctly and appears in octoprint, but octodash is not updated
- start a print from octoprint, print view does not show up in octodash

**Expected behavior**

- hotend, bed and fan values should be updated on octodash screen when they change on the machine
- octodash should show the print screen when a print is running

**General Information:**
 - pi3B+, pi4
 - raspios
 - octodash revision: c26a901
 - OctoPrint Version 1.5.3

UnchartedBull, 2021-03-23:
Known bug on the latest master, will have a look at a fix later.

pciavald, 2021-03-23:
Do you have an ETA ? need help ?

UnchartedBull, 2021-03-23:
Will be fixed eod.

PR #1594 (2021-03-23): fix websocket and config upgrade bug
Fixes #1592
closed 2021-03-23 by UnchartedBull