UnchartedBull/OctoDash #1609: display octoprint errors

pciavald, 2021-03-25:
**Is your feature request related to a problem? Please describe.**

When octoprint has a connection failure, octodash is not currently prepared to help the user fix the error. Currently: 
- octoprint fails to connect to the printer at boot or the connection fails
- octodash thinks the printer has been shutdown by the user and goes to sleep mode

**Describe the solution you'd like**

Octodash should listen for octoprint errors and if an error is detected it should display the error instead of going to sleep mode. 

**Additional context**

We've had several cases of errors and are starting to list them. The last one was appearing in a popup in octoprint, i'm not sure which api endpoint should be used to get that. Any advice ? We can work on this one.

UnchartedBull, 2021-03-28:
https://github.com/UnchartedBull/OctoDash/blob/main/src/app/services/socket/socket.octoprint.service.ts#L261 this file to get the data from the OctoPrint Socket (there are way more states available than the few that OctoDash currently has implemented, but the ERROR status is probably the one you're looking for. Just return a different enum value there then and use this service: https://github.com/UnchartedBull/OctoDash/blob/main/src/app/event.service.ts to navigate to a new page.

UnchartedBull, 2021-07-28:
Can be triggered with Virtual Printer with following command:

`!!DEBUG:send Error:STOP called because of BLTouch error - restart with M999`

PR #2586 (2022-02-22): Action commands and printer notifications
continuation of #1661.

Closes #1624
Closes #1609
closed 2022-02-22 by UnchartedBull