UnchartedBull/OctoDash #1617: specify nodejs and npm versions

pciavald, 2021-03-27:
**Is your feature request related to a problem? Please describe.**

it seems we've encountered errors because we're using npm 6.14.11 and you're probably using npm 7 as the package-lock.json says `lockfileVersion: 2`

**Describe the solution you'd like**

in CONTRIBUTING.md, could you please add the npm and nodejs versions you are using to do the builds ?

closed 2021-04-07 by UnchartedBull

pciavald, 2021-04-07:
I still have a difference from your setup i think, see the difference of package-lock.json in #1610

UnchartedBull, 2021-04-07:
Which os are you using? Different operating systems can cause differences in the package-lock iirc.

pciavald, 2021-04-07:
Fedora 33 workstation

UnchartedBull, 2021-04-07:
I'm using a Mac so that is probably why there are still some differences. I think we can safely ignore those though.

pciavald, 2021-04-07:
until we find out, i think it's best that when you merge a PR that modifies package.json, you run `npm install` and commit package-lock.json before merging

UnchartedBull, 2021-04-07:
Sounds like a plan 👍

commit a0e688a2 (2021-04-07): Update contributing (closes #1617)