UnchartedBull/OctoDash #1618: Feedrate decrease does not issue M220

mpalm0508, 2021-03-27:
**Describe the bug**
When decreasing feed rate a M220 is not sent with the new value.  Increase is fine.  My current build is from the lastest commit as of 3/27/2021

Decrease to 100
    Send: N6356 G1 X58.007 Y132.228 E3.11742*100
    Recv: ok
    Send: N6357 M104 S205*85
    Recv: ok
    Send: N6358 M140 S55*109
    Recv: ok
    Send: N6359 M106 S191*87
    Recv: ok
    Send: N6360 G1 X58.219 Y131.862 E0.01431*110
    Recv: ok

Increase works as expected.  

    Send: N6305 G1 X110.921 Y197.250 E1.76234*85
    Recv: ok
    Send: N6306 M104 S205*81
    Recv: ok
    Send: N6307 M140 S55*103
    Recv: ok
    Send: N6308 M106 S191*83
    Recv: ok
    Send: N6309 M220 S150*88
    Recv: ok
    Send: N6310 G1 X111.497 Y197.250 E0.01955*93

mpalm0508, 2021-03-27:
Looking though the source it looks like this issue is only a problem trying to set feedrate back to 100.  Looking at printer.octoprint.services.ts (starting line 105) if feedrate is 100 it just drops out. Quick fix it to remove the check but not sure the reasoning behind it.  

`  public setFeedrate(feedrate: number): void {
    if (feedrate === 100) {
      return;
    }`

UnchartedBull, 2021-03-28:
I honestly can't tell you why it just returns on 100. I'll check this and make sure to fix this before the next release.

PR #1676 (2021-04-08): fix settings, feedrate, add axis inversion
Fixes #1618
Closes #1662
Fixes #1675
Closes #1098
closed 2021-04-08 by UnchartedBull