UnchartedBull/OctoDash #1632: Newest version of Chromium can not be run at root - causes OctoDash not to run

JeremyProffitt, 2021-04-01:
When system boots, OctoDash will not run, erroring out:

FATAL:electron_main_delegate.cc(254)] Running as root without --no-sandbox is not supported.  See https://crbug.com/638180

When I reviewed this error, it appears Chromium no longer supports being run as root - https://crbug.com/638180

How to recreate:

Download and burn to sd card / usb drive, OctoPi (octopi-buster-armhf-lite-0.18.0) from https://octoprint.org/download/
Boot Raspberry Pi 4 with HDMI display or 7" LCD
login and run:  apt install -y update; apt install -y upgrade
install Octodash & Reboot

Full Error message:

X.Org X Server 1.20.4
X Protocol Version 11, Revision 0
Build Operating System: Linux 5.4.0-54-generic armv8l Raspbian
Current Operating System: Linux octopi 5.10.17-v7+ #1403 SMP Mon Feb 22 11:29:51 GMT 2021 armv7l
Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=d53bd085-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
Build Date: 15 January 2021  02:03:57PM
xorg-server 2:1.20.4-1+rpt2+deb10u3 (https://www.debian.org/support)
Current version of pixman: 0.36.0
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Thu Apr  1 14:49:02 2021
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
[28797:0401/144903.312800:FATAL:electron_main_delegate.cc(254)] Running as root without --no-sandbox is not supported. See https://crbug.com/638180.
Trace/breakpoint trap
xinit: connection to X server lost

waiting for X server to shut down XIO:  fatal IO error 4 (Interrupted system call) on X server ":0"
      after 157 requests (157 known processed) with 0 events remaining.
(II) Server terminated successfully (0). Closing log file.

JeremyProffitt, 2021-04-01:
OctoDash v2.1.2

If I repeat the process, and do not perform a 'sudo install -y update; sudo install -y upgrade' then octodash does work.

douglasdtlltd, 2021-04-03:
Is there a way to downgrade the version chrome without doing a new install?

JeremyProffitt, 2021-04-03:
> Is there a way to downgrade the version chrome without doing a new install?

That's basically what I did when I didn't do the apt update/upgrade.  But  this is a larger issue, as the chromium capabilities will eventually become main stream in img releases, causing the software to fail out of the box.  Locking octodash into a specific version limit of chromium will cause larger issues down the road.

pciavald, 2021-04-05:
@JeremyProffitt could you please detail how you start octodash ?

JeremyProffitt, 2021-04-05:
@pciavald using the option to start on boot in the install script.  That in turn adds the start up to /root/.bashrc

UnchartedBull, 2021-04-06:
v2.1.2 is still running on Electron 9, which is fairly outdated by now (12 is the current version). The beta is already running on Electron 12. I don't know whether that fixes the chromium issue or whether that still persists, but if you have the time and want to test some stuff feel free to try the beta and report back :)

UnchartedBull, 2021-04-06:
Just had a look at the electron repository and they say it's due to Electron being run with the root user (which OctoDash shouldn't do if you've installed everything with the script). Do you executed the script in a superuser context by any chance? The issue you linked is also from 2016 and has been fixed in 2017, so I think there is something else going on here.

JeremyProffitt, 2021-04-06:
The installation script was not run as root, this is validated by the ask of a password during installation.

As for the electron issue, I don't know enough about electron to know, but I'm a senior dev ops engineer, so happy to poke around if you have additional questions.  But this use case, install octoprint, update then install octodash is extremely happy path and failed multiple times on multiple configurations in my lab.

pciavald, 2021-04-06:
@JeremyProffitt could you please post the output of:
```
whoami
uname -a
cat /root/.xinitrc
cat ~/.xinitrc
```

Looking at the [install script](https://github.com/UnchartedBull/OctoDash/blob/main/scripts/install.sh#L757), if it had been ran as a regular user, it should not have written to `/root/.xinitrc` while evaluating `~/.xinitrc`

UnchartedBull, 2021-04-08:
Oh I totally over read that, if OctoDash is started from the `/root/.bashrc` file then it will be run in a superuser context, which will, correctly, throw the error you mentioned. So @pciavald is on the right track here. Please provide the requested information. I also haven't heard of any other people experiencing this problem so far ...

UnchartedBull, 2021-05-03:
Closing since there were no updates. Please reopen with the requested information if this is still valid.

closed 2021-05-03 by UnchartedBull