UnchartedBull/OctoDash #1647: Installation failed

kekko7072, 2021-04-04:
**What doesn't work?**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**What did you already try?**
A clear and concise description of what you tried to make OctoDash work.

**General Information:**
 - Hardware [e.g. Raspberry Pi, if you have layout issues please also include your screen resolution]
 - OS Info [e.g. Raspbian Buster Lite, please also indicate if you used OctoPi]
 - OctoDash Version [e.g. v1.0.0]
 - OctoPrint Version [e.g. v1.0.0]

**Additional context**
Add any other context or screenshots about the feature request here.

kekko7072, 2021-04-04:
https://github.com/UnchartedBull/OctoDash/issues/1388#issuecomment-809548547

kekko7072, 2021-04-04:
[image]

kekko7072, 2021-04-04:
I tried all the guide online but nothing works, I use a completely fresh distro of octopi but with no results.

tobsiyo, 2021-04-04:
Looks like it's the same problem like i had. Octodash isn't compatible with this RPi -> illegal instruction.

kekko7072, 2021-04-05:
> Looks like it's the same problem like i had. Octodash isn't compatible with this RPi -> illegal instruction.

Did you changed monitor? So what monitor fidi you buy. @UnchartedBull this should be added to the info of the project that RPi monitor are incompatible.

tobsiyo, 2021-04-05:
Your Raspberry Pi, not your Screen.

kekko7072, 2021-04-05:
But it's new? What should I buy? It pi 3 model b

pciavald, 2021-04-05:
@kekko7072 can you please detail how you installed octodash ? illegal instructions looks like you're trying to run the wrong version of octoprint, ie not the armv7l version.

kekko7072, 2021-04-05:
I downloaded from website, installed, logged in, enabled in settings auto login, run apt upgrade & update the install octodash using command. then to fix I tried to install x init and x server

kekko7072, 2021-04-05:
If you suggest me step by step I retry. Reilly glad to be helped

pciavald, 2021-04-06:
@kekko7072 can you please post the output of `uname -a` ?

kekko7072, 2021-04-06:
> @kekko7072 can you please post the output of `uname -a` ?
Yes ``Linux octopi 5.10.17+ #1403 Mon Feb 22 11:26:13 GMT 2021 armv6l GNU/Linux Linux octopi 5.10.17+ #1403 Mon Feb 22 11:26:13 GMT 2021 armv6l GNU/Linux Linux octopi 5.10.17+ #1403 Mon Feb 22 11:26:13 GMT 2021 armv6l GNU/Linux ``

pciavald, 2021-04-06:
@UnchartedBull can confirm but i think there's no automated build for your architecture, `armv6l` (i'm guessing pi zero), so the installer [downloaded and installed](https://github.com/UnchartedBull/OctoDash/blob/main/scripts/install.sh#L676) the armv7l version instead.

After a quick research it seems electron, the base framework of octodash, does not offer builds for armv6 unfortunately. It seems OctoDash can thus not be installed on raspberry pi zero.

pciavald, 2021-04-06:
I forgot you mentioned that you're using `Raspberry Pi 3 model B`, here's the list of armv6 devices that are incompatible while electron can't build for armv6:

[image]

kekko7072, 2021-04-06:
> @UnchartedBull can confirm but i think there's no automated build for your architecture, `armv6l` (i'm guessing pi zero), so the installer [downloaded and installed](https://github.com/UnchartedBull/OctoDash/blob/main/scripts/install.sh#L676) the armv7l version instead.
> 
> 
> 
> After a quick research it seems electron, the base framework of octodash, does not offer builds for armv6 unfortunately. It seems OctoDash thus can not be installed on armv6 devices.
> 
> 
> 
> Here's a list of raspberry pi armv6 devices:
> 
> [image]
> 
> 
> 
> You mention that you use raspberry 3B, are you sure about the model? It's written on the top of the board. The table seems to tell that Raspberry 3B should be armv8.
> 
> 

Yes it is a Raspberry pi model B+ v1.2

[image]

pciavald, 2021-04-06:
Your board is a `RPi 1 Model B+` armv6 from 2014, not a `RPi 2 Model B v1.2` nor a `RPi 3 Model B` from 2016.

I'm sorry, but i think OctoDash is currently incompatible with your hardware. It should probably be noted in the documentation.

kekko7072, 2021-04-06:
Now I understand why a month ago I also tried installing on raspberry pi zero and didn't work. In fact it is not officially supported by octopi but it works (the problem is when you use webcam stream), but as you show me is not compatible with electron and so with OctoDash, correct? 

What should I do with Raspberry Pi Model B +?
Should I purchase another hardware to make it work or try reinstall?

pciavald, 2021-04-06:
The incompatibility is in that OctoDash uses Electron to create the window that shows the app, and Electron can build for most platforms, but it seems not for armv6. Any device on the table that is armv7 or armv8 will work.

kekko7072, 2021-04-06:
Ok clear thank you so much for the patience and knowledge. I will purchase the new hardware. Witch do you think is better? This issue can be closed.

pciavald, 2021-04-06:
I think it should be documented before closing.

RPi 4 is the best of the line currently, RPi 3B+ will do very fine too.

UnchartedBull, 2021-04-06:
Was already in the wiki (second paragraph): https://github.com/UnchartedBull/OctoDash/wiki/Supported-Devices. Just updated the text a little to make it more clear.

closed 2021-04-06 by UnchartedBull