UnchartedBull/OctoDash #1653: Communication errors / dns etc.

hawkeyexp, 2021-04-05:
i was running octoprint server 1.5.3 and on second pi octodash nightly from 02-27-2021 which was running nice.
Then i switched to latest 04-04-2021 and result was no communication.
I startet to clear the octoash system to reconfigure but no change - octopi host not detected.

ok starting from scratch - clean image and again build 04-04-2021 - partital success - after a lot of tries i got the api request in the octoprint server gui - acccept and finished setup.
But plugins will also fail to communicate.

ok giving octoprint server 1.6.0rc1 a try - the next fail - seems octodash has a problem with version string which contains a-z - i got an error message while searching for hosts.

I also tried to manually create an api key and setup config.json manually - then i get an endless leep trying to connect with splash screen - it makes no differ if i use dns or ip.

I'm fully frustrated and no idea what is going on - all dns and ip checks for communication are working without errors.

Any ideas ?

Reinstall octoprint server would be last part but for sure i won't do that without understanding what's going on.

UnchartedBull, 2021-04-06:
OctoDash currently doesn't read the OctoDash version string, so this shouldn't affect operation at all. Did you try to connect to `octopi.local` or `localhost`? Did you try and enabling CORS like mentioned here: https://github.com/UnchartedBull/OctoDash/wiki/Troubleshooting#octodash-cant-connect-to-octoprint-0-unknown?

scottmudge, 2021-04-06:
Having the same problem. Just updated to 04-04-2021 from 02-27-2021, and it stopped connecting.

It's just stuck at "Initializing...". I have CORS enabled, set it to "localhost", used the correct API key. It was working just before updating, and after updating stopped working. Double checked the configuration. 

Let me know if you'd like any logs, but for now I need to downgrade.

scottmudge, 2021-04-06:
Alright switching from:

`"url": "http://localhost:80/api/",`

to 

`"url": "http://octopi.local:80/",`

in the config seemed to fix it. I suppose it was the `/API` bit that was causing problems.

hawkeyexp, 2021-04-06:
I managed to get my install running again - i started fully from scratch with plain raspbian lite to install octodash but i had to understand it is only possible if octoprint is also installed on same mashine - ok also done manually.
After install i tried to connect but also failed (localhost, too).
Ok rebooting all my network stuff like router and 1 repeater which is used as wifi client to lan - after this step i could connect again - it seems there was an issue with dns resolving/caching.

@scottmudge Just funny: my install is running with "http://octopi.local:80/api/" which is an external host :-)

@UnchartedBull: it seems the system is primary using dns/avahi - is this correct? If yes and if possible you should ignore dns stuff if an ip adress is given.
I could be fully wrong but with my observations it seems so :-)

ps: CORS seems to be not part of the problem :-)

UnchartedBull, 2021-04-08:
@scottmudge the `/api` part got removed from the config for v3.0.0 onwards. This normally should've been done automatically, but you may have encountered a corner case where you selected just the right dev builds to skip the migration :D

@hawkeyexp
- OctoPrint doesn't need to run on the same machine as OctoDash, as long as OctoDash can access OctoPrint via the network everything should work just fine.
- octopi.local isn't an external host in all cases, it might be pointing to the local device. The main problem here is that OctoPrint is only advertising the local dns name via avahi / bonjour. This caused a couple of problems so far, so in v3 OctoDash will actually check during setup whether the advertised mDNS name is actually pointing to localhost. 
- If an ip address is given OctoDash should ignore all the DNS stuff.

Closing this here since everything seems to be working again. If any of the problems reoccur please reopen.

closed 2021-04-08 by UnchartedBull