UnchartedBull/OctoDash #1659: Logging

UnchartedBull, 2021-04-07:
Enable (optional) verbose logging for OctoDash. Due to the nature of OctoDash this requires some workarounds:

Current idea (very basic, open to changes):

- Use [winston](https://www.npmjs.com/package/winston#adding-custom-transports) and log to a file
  - I've had good experiences with winston, so this would be my go to logger
  - Can define a max filesize (I think something like 100kB might be sufficient)
- Create a custom logger service for the frontend
  - This service should push all looks to Electron through the IPC, which winston will then save to the file
- Write a simple express server which serves the file, maybe alongside a download button
  -  The server should be controllable via the settings menu (as well as the verbose logging)

There are still a few open questions:
- How to handle errors that are directly logged to the Chrome console (without any explicit log statement)
- Maybe use a "lighter" webserver, since it would only serve a single website

pciavald, 2021-04-08:
By the time this gets decided, we can easily search and replace all "console" statements :)

UnchartedBull, 2021-04-08:
console.log might be fine for the electron app, but we should keep it out of the Angular part for now. Chromium doesn't garbage collect anything that is logged to the console and since OctoDash is never really refreshed this could cause (and has caused) a memory leak. So keep the log statements in the electron part, but please don't add any in the frontend part.

pciavald, 2021-04-08:
understood!

cryptoAlgorithm, 2022-07-02:
@pciavald The approach some huge production apps take is to override the console object and proxy all calls to log debug etc with ones that also log to wherever else that's required. In this case it could be redirecting the calls to equivalent Winston calls. 

@UnchartedBull If you're planning to log to disk, I think it would be a good move to log to a Ramdisk so as to not kill the sd card.