UnchartedBull/OctoDash #1677: let octoprint manage printer differences

pciavald, 2021-04-09:
This feature emerged from discussion on [this commit](https://github.com/UnchartedBull/OctoDash/commit/39e3c168841f2fa60c51eb7755ee121371301295).

**Is your feature request related to a problem? Please describe.**

When pressing "back" on the "babystep adjust" screen, `M500` is sent to the printer to save the Z-offset value to EEPROM in Marlin. However, Klipper does not support this command and will just return "unknown command" without hard-saving the value, which should be done by `SAVE_CONFIG` which resets the board, and is not acceptable in-print.

**Describe the solution you'd like**

All manual gcode commands should be removed from octodash, to use octoprint's APIs instead. All machine configuration should be handled in octoprint to let octodash focus on its own task: be the best dashboard, regardless of the printer in use.

**Additional context**

I'm opening this issue to enable debate on which features are hard-coded printer behaviors in octodash. I've already noted in the comments of the commit that at least 2 marlin commands are defined in the user settings:

```json
  "disableExtruderGCode": "M18 E",
  "zBabystepGCode": "M290 Z",
```

cp2004, 2021-04-19:
> All manual gcode commands should be removed from octodash, to use octoprint's APIs instead. All machine configuration should be handled in octoprint to let octodash focus on its own task: be the best dashboard, regardless of the printer in use.

There's nowhere that these gcode commands are defined in OctoPrint, you have to define them somewhere first. Can't configure them in OctoPrint out of the box afaik.

UnchartedBull, 2021-04-20:
@cp2004 is correct here. Not all functionality is available from Octoprint, i.e. Babystepping ist not available AFAIK. This should be solved by the new architecture though. 

All services that are communicating with OctoPrint are behind an abstract service now. This means they can be easily exchanged. The way forward here would be to create either a native Klipper-Service or a Klipper flavoured OctoPrint service.

pciavald, 2021-04-20:
About Klipper, we have planned to test it but did not yet so i can't really talk about it, but as far as I understand there's some support already in place to use it with octoprint.

To me it would make sense to create at least a meta-package for octoprint called octodash or something, that would have the package dependencies that octodash needs, for instance displaylayerprogress. The `octodash` octoprint package could have its own behavior like enabling CORS, but most of its use should probably be already done by already existing packages, maybe installing configurations for it.

This way we could provide ourselves with the APIs we need in octoprint by only updating this meta-package.

UnchartedBull, 2021-05-03:
There is a plugin available to configure OctoDash via OctoPrint. The main point here is that OctoDash should be able to run with Klipper without any OctoPrint installed at all, thus it wouldn't make sense to introduce a plugin now that OctoDash relies on.

pciavald, 2021-06-03:
keeping this here for reference https://github.com/jneilliii/OctoPrint-OctoDashCompanion