UnchartedBull/OctoDash #1700: Moonraker + Klipper + Octodash? Is it possible?

alexcroce, 2021-04-16:
Hello, i use klipper on my printer, and i have moonraker + klipper on my raspberry. Can i install Octodash and get it working on my 5" touchscreen connected via hdmi to my raspberry?
If yes, i need to manually install it?

cp2004, 2021-04-19:
Quoting the front page of the project:
> OctoDash is a simple, but beautiful dashboard for **OctoPrint**. Please read the instructions carefully!
OctoDash is a User Interface for **OctoPrint**, it utilizes the **OctoPrint** API

If you are using OctoPrint, then it will work. If you are not using OctoPrint, then of course it will not work.

UnchartedBull, 2021-04-20:
not possible atm as @cp2004 perfectly explained :). This might (or might not) change in the future. There have been discussions on a native Klipper integration, this will take at least a couple of months though.

closed 2021-04-20 by UnchartedBull