UnchartedBull/OctoDash #1702: BTT TFT70 , Octodash doesn't fill the screen

Killajoedotcom, 2021-04-17:
**What doesn't work?**
I have a BTT TFT70 w/ a Pi 3B+ running Octoprint and OctoDash. Currently the OctoDash doesn't fill the space of the TFT leaving about 25% of the right side of the screen unused. Also, the touch buttons are misplaced. Seems like a scaling issue. 

**What did you already try?**
I have tried updated the raspi-config to account for the large display. No joy.

**General Information:**
 - Hardware [Raspberry Pi 3B+ Screen resolution ]
 - OS Info [OctoPi 0.18.0]
 - OctoDash Version [e.g. v2.1.2]
 - OctoPrint Version [e.g. v1.5.3]

**Additional context**
Add any other context or screenshots about the feature request here.

UnchartedBull, 2021-04-19:
Please provide a photo of how your screen looks. Most likely something misconfigured. OctoDash scales fine up to 4k resolution.

Killajoedotcom, 2021-04-21:
Here is a screenshot of what I'm seeing. Not only is it the scaling error, but another connection error as well.

[image]

pciavald, 2021-04-21:
Seems like a pretty bad hardware problem to me. Maybe try with a properly working screen.

Killajoedotcom, 2021-04-22:
I really don't think it's a hardware problem. changing certain settings improved the picture, so it has to be that. Anyone have ideas?

UnchartedBull, 2021-04-23:
This definitely is either a bad screen / a badly configured Raspbian installation. Please try a clean install with Raspbian Desktop (maybe on a separate SD card) and check if the artifacts are still visible on the Desktop without OctoPrint / OctoDash installed.

QC can be a bit funky in China so there might as well be a good chance that your screen is just broken.

Killajoedotcom, 2021-04-24:
SOLVED: Definitely not a bad screen. Here's the solution:

1. wipe the RPi SD card and install a fresh Octoprint image
2. Once completed, update the octopi-wpa-supplicant.txt file on the SD card
3. Update the config.txt file with the following:
[image]
4. Place SD card in the Pi and login via SSH
5. Run the Octodash install command: bash <(wget -qO- https://github.com/UnchartedBull/OctoDash/raw/main/scripts/install.sh)
6. Reboot and profit 

Thanks for the help folks.

closed 2021-04-24 by Killajoedotcom