UnchartedBull/OctoDash #1703: Adding Support for TP-Link Smartplug and Spool Manager

douglasdtlltd, 2021-04-19:
Use TP-Link Smartplugs and Spool Manager for more features & easier interface. 

Replace PSU control and Filament manager from 2017 with newer apps and plug-ins.

https://plugins.octoprint.org/plugins/tplinksmartplug/

https://plugins.octoprint.org/plugins/SpoolManager/

I feel like both of these are better and more accessible than their counter parts. You dont need a strange PSU or solenoids, just a wifi plug. And I had no problem configuring Spool Manager.  I had had trouble figuring out why I wasn't allowed to type in fields, and Filament manager wanted a database set up. Spool manager didnt need that. And if it did, it did it on its own.

UnchartedBull, 2021-04-19:
Just because they're old doesn't mean that they stop working. PSUControl is still widely popular (more installations than TPLinkSmartPlug), so replacing some existing functionality won't happen.

Nevertheless: Support for TPLinkSmartPlug already has been added and will be released with v3.0.0. SpoolManager might take a little longer, since there is no proper API available. duplicate of #667.

closed 2021-04-19 by UnchartedBull

douglasdtlltd, 2021-04-19:
When will the update be ready? I dont see a way that PSU control can turn off or on a wifi outlet? Also how is my post/request a duplicate, i just see "@angular/language-service"l or am i confused?

UnchartedBull, 2021-04-19:
Probably in the next days. The beta is already available in the discord if you can't wait :).

Sorry linked the wrong issue. I updated the comment and the correct issue has been linked.

kantlivelong, 2021-04-20:
> I dont see a way that PSU control can turn off or on a wifi outlet? 

Sure it can.

UnchartedBull, 2021-04-20:
Yes Tasmota and Tasmota-MQTT will also be included thanks to you @jneilliii :)