UnchartedBull/OctoDash #1704: UCTRONICS Touchscreen doesn't work

ryryog25, 2021-04-19:
**What doesn't work?**
I have installed the drivers for my UCTRONICS LCD35 HDMI RPI screen and I cannot use the touchscreen, it just doesn't respond. I can control the screen with VNC, so I know that OctoDash is working, I just cannot do anything on the screen itself!

**What did you already try?**

-  I have tried using the drivers labeled UCTRONICS LCD35 RPI, both the normal and HDMI versions. Pretty much every driver from UCTRONICS that was similar to my screen that I could find.
- Remove VNC to make sure that wasn't the issue
- Uninstall/Reinstall OctoDash
- Uninstall/Reinstall Touch Drivers (using the included backup script before install and restore script.)

**General Information:**
 - Raspberry Pi 3B+
 - Octopi with updated dependencies, etc.
 - OctoDash Version v2.1.2
 - OctoPrint Version v1.5.3

**Additional context**
This was a pretty much clean install... I had flashed octopi, installed and restored Octoprint and then installed klipper. Haven't done a whole lot (or really anything) to system config. I did notice though that for a couple of seconds after boot I can in fact touch and interact with OctoDash and then it seems to freeze up so it seems like the issue isn't my drivers, but rather something with OctoDash.

UnchartedBull, 2021-04-20:
This is most likely a driver issue, at least it has been for all other similar issues. Please search the closed issues and ask Google (I don't own that screen, so I can't provide any support). I think it was a different driver that solved it for most people.

Please also try a different app and check if everything is working there. If you're sure that this issue is caused by OctoDash please reopen.

closed 2021-04-20 by UnchartedBull

UnchartedBull, 2021-04-20:
Oh this might also be helpfu: https://github.com/UnchartedBull/OctoDash/wiki/User-Guides

ryryog25, 2021-04-20:
I am not having a driver issue... OctoScreen and TouchUI, as well as the desktop all worked fine beforehand. However a few seconds after OctoScreen starts, it becomes unresponsive.

ryryog25, 2021-04-20:
I actually ended up figuring out the other day that this is an issue with an Octoprint plugin because when Octoprint crashed during my testing and rebooted into safe mode, the screen worked fine. The culprit ended up being none other than "Filament Sensor Reloaded" likely due to the way it interfaces with the GPIO pins, which is how the touch portion of the screen connected, while the rest came through HDMI. (This would also explain why my PSU relay has been iffy.)

ryryog25, 2021-04-20:
I have now removed the plugin, it was deprecated anyways, and am using "Filament Sensor Simplified" which seems to work well and does not interfere with GPIO connections.