UnchartedBull/OctoDash #1705: tft not displaying octodash, only displaying through HDMI

boannarrow, 2021-04-19:
I'm new to programming and having trouble getting my touch screen to display Octodash, it displays when i have an HDMI plugged in but my screen doesn't have an HDMI port so I'm only using gpio pins. I've been looking through the xorg files and config.txt to see how I can make it default to the touchscreen but I don't see anywhere specifying the lcd settings. The touchscreen is displaying command line not Octodash. I have an adafruit 3.2" display and rpi 3A+ using octopi 0.18.0 and octoprint 1.5.3

Will-wastelander, 2021-04-20:
This is a driver or xorg issue most likely, and not an issue with OctoDash. You will most likely need to change the terminal used for the xorg display, if I remember correctly.

boannarrow, 2021-04-20:
Ok, how would I go about that? Really new at this.

Will-wastelander, 2021-04-25:
It should be related to this line. Try google searching for xinitrc terminal or something like that.

https://github.com/UnchartedBull/OctoDash/blob/661db3f02b32307f3fd7c7c1e43d7edad38e5f10/scripts/install.sh#L780

Will-wastelander, 2021-04-25:
This might help also.

https://github.com/UnchartedBull/OctoDash/issues/1582#issuecomment-803433457

closed 2021-06-05 by stale[bot]