UnchartedBull/OctoDash #1725: How to update to get the language files

GuillaumeCyriaque, 2021-04-25:
Hello,

How can I update OctoDash with the language files? I tried a manual update (2.1.2) but it seems that these files were not installed? Have I forgotten something?

General Information:
 - Hardware: Raspberry Pi 4
 - OctoDash Version 2.1.2
 - OctoPrint - OctoPi Version 0.18.0
 - My location is defined on fr_FR.UTF-8, my OS is in french.

Thanks

pciavald, 2021-04-27:
Hi,

The multilanguage feature will be released in v3.0.0. If you want to give it a try before, you can clone and build the main branch of this repo.

closed 2021-04-27 by pciavald

GCYR3278, 2021-05-01:
OK, I misunderstood.  I will wait for v3.  I've only been using a raspberry for 2 weeks, I don't have the skills to do more yet now. Thanks.