UnchartedBull/OctoDash #1727: Use a default config scheme to allow a minimal config file

pciavald, 2021-04-27:
Opening this issue to debate an idea that emerged from previous comments on PRs.

I think it's good practice to have default configuration of the app to allow minimal configuration files. Currently the config services in the main process and renderer check the config file for key completeness. I think checking the config for validity is a great feature, but i think missing keys should be replaced by their default value, as i had done for the extruder control. This allows users to have a very small and readable config file with only the modifications they need instead of a redundant information store (as configs already do have defaults in the code, that are overridden at all times by the defaults we write in the config file).

Not a critical issue, but i had it in mind and it's pretty easy to implement.

UnchartedBull, 2021-04-28:
Not a huge fan of this as this has two implications that aren't that great from my side:

- Default values will be spread around the whole code-base (currently the default are all stored in `config.default.ts`)
- While the config may be more readable it would be harder to edit (from my point of view). If you'd like to add something, which isn't included in your config yet you need to open the GitHub Wiki, search for the right JSON attributes and place them in your config. If all your values are already there you can just update the value, without having to open any external site (granted that the type can be deferred).

So I'm not in favour of this unless it has some other benefits :)

pciavald, 2021-04-28:
As I said it's not a critical issue so can be closed in case it's not relevant, but I like to discuss different ways of doing stuff :)

1. I don't see why we would change the current defaults stored in `config.default.ts`, I think it's very clean this way. Why did you think that was what I suggested ?
2. The way i was picturing it was we ship the configuration with all options populated to defaults, so that the config file is its own documentation (could be something else than JSON by the way, to allow comments, like YAML, TOML or even ini), but removing everything in the config file would not invalidate it, just use the defaults in `config.defaults.ts`.

Basically the only change i'm suggesting is to allow any config entry to be undefined, taking the default in this case.

UnchartedBull, 2021-05-03:
I don't really see the need to remove entries from the config. If we pre-populate everything why should a user go through the config file and delete stuff that seems unnecessary. By now OctoDash has a fairly good config upgrade system to keep all configs on the same state, so I don't see the need for minimal configs, this just seems like a lot of `null` checks in the config-service.

closed 2021-05-03 by UnchartedBull