UnchartedBull/OctoDash #1749: NOX theme Layer progess wrong screen position and icon missing

ipa64, 2021-05-01:
Hi,
In this case pictures are better than world.
https://i84.servimg.com/u/f84/20/23/21/77/captur13.png
https://i.servimg.com/u/f84/20/23/21/77/captur13.png

Thank you

ipa64, 2021-05-01:
Tested with theGarbz/Focus same behavior

ipa64, 2021-05-01:
until v2.1.1 app was layer-progress
from v2.1.2 app is height-progress

UnchartedBull, 2021-05-03:
Please add the requested information in the issue template. Otherwise no one can help you.

pciavald, 2021-05-03:
I'm getting the same issue, NOX theme maintainer knows about it (I can't remember which issue it was in) and is already fixing it. @ipa64 you can fix it yourself before by fiddling with the custom css. Something like this
```css
.height-indication {
  width: auto !important;
  position: fixed !important;
  top: 52vh !important;
  left: 59vw !important;
}
```

ipa64, 2021-05-04:
> I'm getting the same issue, NOX theme maintainer knows about it (I can't remember which issue it was in) and is already fixing it. @ipa64 you can fix it yourself before by fiddling with the custom [...]

Thank you pciavald, 
I have already tried that without success (and many variants😂), for now my workaraound is to move the line with temperatures a bit higher and the fan line a bit lower.

thegarbz, 2021-05-14:
@ipa64 What screen resolution are you using? You said you had the problem with my focus theme as well, if I can replicate it somehow I'll try and sort it out. But right now I'm unable to see the problem. 
Can you also add a screenshot from my theme? It's curious that you're not only loading in the wrong position but you're not loading the icon...

ipa64, 2021-05-15:
@thegarbz, I am using official raspberry touchscreen 7", resolution 800x480.
On my first post you should find 2 print screens, maybe they was made from NOX theme(I forgot), but it’s the same behavior. 
Now I am using your focus theme, my workaround was to move a bit higher the temperature line, and move a bit lower the fan line, to make place for the layer progress line.
Here the result: https://i84.servimg.com/u/f84/20/23/21/77/e6db1310.jpg
Thank you

ipa64, 2021-05-16:
@thegarbz , this morning I reinstalled original focus custom-styles.css, and got some screenshots from vnc.
https://i84.servimg.com/u/f84/20/23/21/77/captur14.png
https://i84.servimg.com/u/f84/20/23/21/77/captur15.png

After some comparaison from previous version of octodash, I noticed there is some deep modifications for layer progress in 2.1.2, thru 2.1.1 that was layer-progress from 2.1.2 it's height-progress app.

thegarbz, 2021-05-17:
@ipa64 I'll take a look at it this week. To confirm what versions of everything are you running? 
Octodash
Octopi
and while I don't think it's relevant the DisplayLayerProgress plugin as well?

ipa64, 2021-05-17:
@thegarbz 
Octoprint 1.6.1
Octodash 2.1.2
Octopi 0.18.0
DisplayLayerProgress 1.26.0
Thank you

thegarbz, 2021-05-17:
@ipa64 for the theGarbz/Focus theme adding the following in the section under ******PRINTING****** will display the Layers in the correct place. 
I dug through the code today and the recent change is missing a class needed to make the Layer icon appear. I've added a PR for that to be fixed in the next version of Octodash but in the meantime at least this will place the Layer text in the correct place and not overlap it on other elements:

`app-height-progress .height-indication {
  position: absolute !important;
  top: 60vh;
  left: 68.8vw;
  width: auto !important;
  margin: 0 !important;
  color: var(--dark-text);
  /*visibility: hidden;
  letter-spacing: -1em;*/
}

app-height-progress .height-indication__current-height {
  position: relative;
  top: -0.5vh !important;
  visibility: visible !important;
  font-size: 5vw !important;
  letter-spacing: normal !important;
  color: var(--light-text);
}`

thegarbz, 2021-05-20:
@ipa64 both the NOX theme and the Focus theme have been updated. If it's all working you should be able to close this issue now.

closed 2021-05-20 by UnchartedBull

ipa64, 2021-05-20:
Now all is working perfectly 
Thank you