UnchartedBull/OctoDash #1752: Display motors position

pciavald, 2021-05-03:
**Is your feature request related to a problem? Please describe.**

Currently there's no way to see or adjust the motors position (X, Y, Z positions).

**Describe the solution you'd like**

Show motors position. I'm not sure where it would be best to show it currently, I would say the controls but it's already quite full of info, it could be a new screen or we could collapse quick actions on the controls screen ?

UnchartedBull, 2021-05-03:
What would be the advantage of knowing the motor positions? This does seem like a very advanced feature (I haven't looked once at motor position, nor do I know where to find those) and probably is way out of scope for OctoDash

pciavald, 2021-05-03:
I've double-checked the reason with my associate and that is because our clients are going to need it to make custom gcodes for part ejection, which is indeed an advanced feature. Also prusa does show it on screen, so there may be other (probably debug) use cases for it. Can be closed if out of scope for upstream.

closed 2021-05-03 by UnchartedBull