UnchartedBull/OctoDash #1753: Access controls during pause

pciavald, 2021-05-03:
**Is your feature request related to a problem? Please describe.**

When pausing a print, it's currently impossible to manually control the printer. During a print, we had to cancel the print to solve a clogged head, which could have been done during the pause.

**Describe the solution you'd like**

Allow access to print controls on the pause screen

UnchartedBull, 2021-05-03:
duplicate of #828

closed 2021-05-03 by UnchartedBull

pciavald, 2021-05-03:
Can you help me understand how accessing the controls from the pause screen is related to the mentioned issue, that seems more about having more control on the control screen's buttons ?

UnchartedBull, 2021-05-03:
You would have full access to the control screen while paused (so the arrow to move the tool around and all Custom Actions). This is all the manual controls OctoDash does offer.