UnchartedBull/OctoDash #1758: Auto webcam view during first x layers for nozzlecam

hawkeyexp, 2021-05-04:
Some time ago i installed a nozzlecam and it would be nice if octodash would be able to show the nozzlecam during first layers to monitor if all is printed well.

My idea: split the print dialog in 2 parts: a left cell with the cam image (2/3 of screen size) and a right one with the shorts info's about temp etc.

Requirements:
- a setting which defines how many layers are shown with cam before switching back to default print dialog
- url for webcam which should be shown

Would be nice to have a feature like this :-)

jneilliii, 2021-05-04:
I just added a static camera page to the octodash companion plugin that will load in octodash using custom actions, if you're running octoprint and octodash on same server. Details in the release notes.

hawkeyexp, 2021-05-04:
You are talking about the iframe integration for a camera stream right? This way i know but custom actions are niot available during the print.

jneilliii, 2021-05-05:
🤦 right, my bad. That definitely would be a stumbling block.

hawkeyexp, 2021-05-05:
No problem @jneilliii :-) The requerst i think is not a big deal i think cause the webcam feature is still included (which is nice)  - it's only the missing part to reach it via the print dialog - currently it's only usable while not printing :-)