UnchartedBull/OctoDash #1779: Display Off Function

stikako, 2021-05-08:
Is it possible to add a custom action to turn off the display? This way it can be off while printing, tap to wake. 

Or even have it set to "timeout" after a period of time, which could be set by user. 

Off, 5 min, 30 min, 1 hour, perhaps?

And then touch the screen to wake it. Might help prolong the life of the screen. 

I requested this function with the Octodash Companion plug-in, however he mentioned Display off is not one of the custom actions. So even if it isn't a part of the main setting, if its one of the actions, it could be added as a button in Octodash Companion.

kantlivelong, 2021-05-12:
Just wanted to chime in on this as well. It seems like the companion plugin should become an official requirement and act as a shim between OctoDash and OctoPrint to bring functionality like this.

ffutlapuztar, 2023-01-29:
is there perhaps a workaround to switch off display manually via custom command and switch on via touch- and everything while he printer is connected to the pi?