UnchartedBull/OctoDash #1781: Disable scrollbars for iframes

hawkeyexp, 2021-05-09:
it would be nice of you can disable the scrollbars for iframes - for example my cam image is shown with a vetical scrollbar - i tried to resize my cam image but no change - i think generally disabling scollbars is ok for embedded content :-)

jneilliii, 2021-05-09:
I know you ar eusing a remote OctoDash Install, but if you have OctoDash Companion plugin installed it includes a webcam view that will auto-size your camera to fit the page, so you don't get scrollbars at all. You just have to use the ifram address `http://ip or name of pi/plugin/octodashcompanion/webcam`

hawkeyexp, 2021-05-09:
You are right - it could be a solution if you implement multicam support :-) I wan't to use my nozzle cam which is the second one not the default :-)

jneilliii, 2021-05-09:
What is the other plugin that you use to enable multicam support within the OctoPrint UI. I could detect that and allow for selecting the stream possibly...feel free to open a Feature Request on the plugin repo for that.

UnchartedBull, 2021-05-18:
This might break the integration for some people that rely on the scrolling features. For camera views this will be tracked in #918 (which should offer a better experience than opening the mjpeg website and no scrollbars), a general option to disable scrollbars won't be useful IMHO.

closed 2021-05-18 by UnchartedBull