UnchartedBull/OctoDash #1807: Rewrite pause and resume commands

seacred, 2021-05-15:
Hi,
as im using klipper pause and resume are called by a gcode macro.
Is there any possibility to rewrite the pause und resume command send via the touchscreen to send my gcode macros instead the regular resume call to octoprint?
I would appreciate any kind of help.
Thank you!

jneilliii, 2021-05-15:
couldn't you add whatever klipper command you want to OctoPrint's pause/resume GCODE Scripts?

seacred, 2021-05-15:
Sadly i can't because i need to store my current position in the macro. 
Klipper needs to use this variables to resume at the correct position. Octoprint isn't able to do this.
i surely can provide the macro if needed.

jneilliii, 2021-05-15:
It might be possible with these examples...you would just have to adjust for the macro.

https://docs.octoprint.org/en/master/features/gcode_scripts.html#more-nifty-pause-and-resume

seacred, 2021-05-15:
No need to implement that, if it's easier to modify octodash to send "macro_resume" instead of octoprints "resume".

UnchartedBull, 2021-05-18:
OctoDash uses the OctoPrint API for pausing / resuming: https://github.com/UnchartedBull/OctoDash/blob/9f702ae84ee49c59c22aabf63d463284bdca7e2b/src/app/services/job/job.octoprint.service.ts#L33. I guess you can adjust those lines to your liking (i.e. with `macro_resume`) and then build everything and install. I would also vote for the custom pause / resume scripts though.

closed 2021-05-18 by UnchartedBull