UnchartedBull/OctoDash #1819: WLED Support/Integration

NovaViper, 2021-05-18:
**Is your feature request related to a problem? Please describe.**
Currently, OctoDash seems to only support neopixels that are directly connected to the raspberry pi. With the recent addition of the WLED Octoprint plugin, I've gone and setup WLED with my neopixels (and it's much easier and smoother to setup than directly connecting them to the pi).

**Describe the solution you'd like**
I would like to see some sort of integration with WLED directly or integration of the Octoprint WLED plugin

**Link other projects**
Octoprint-WLED plugin: https://github.com/cp2004/OctoPrint-WLED
WLED: https://github.com/Aircoookie/WLED

cp2004, 2021-05-18:
In the next release of the WLED plugin, there will be @ commands available, so that custom buttons can be created. For my WS281x plugin, these are already implemented for LED control. Not sure about how other plugins manage this, but for mine I prefer @ commands since it means that other people don't need to implement specific support for my (likely to change) API, and for each plugin.

More: https://github.com/cp2004/OctoPrint-WLED/issues/28

If you want to try it out early, let me know in the issue and I can take a dev release to trigger the build.

UnchartedBull, 2021-05-20:
Those @ commands are the way to go from my point of view as well :) Closing this here then.