UnchartedBull/OctoDash #1821: [2.2.0] Layer indicator layout issue during printing/loading of file with Focus Theme

Its-my-right, 2021-05-18:
**Describe the bug**
The Layer indicator is written over other informations with Focus Theme (But it is OK with the default theme)

**To Reproduce**
Steps to reproduce the behavior:
1. Install Focus Theme bundled with 2.2.0 release
2. Load a file or print a file

**Expected behavior**
The layer indicator should appear with an icon under the build plate temperature

**Screenshots**
With Focus Theme:
[image]

With Default Theme:
[image]

**General Information:**
 - Hardware : Raspberry Pi 3B (screen resolution : 800x480)
 - OS Info : Octopi v0.17.0
 - OctoDash Version : v2.2.0
 - OctoPrint Version : v1.6.1

**Additional context**
The issue didn't exist in the 2.1.2 release with Focus theme

Its-my-right, 2021-05-18:
Sorry I didn't see issue https://github.com/UnchartedBull/OctoDash/issues/1749 which looks similar

closed 2021-05-18 by Its-my-right

thegarbz, 2021-05-19:
@Its-my-right FYI a fix is underway but in order to get the old functionality back some changes are required, changes which I submitted too late for the 2.2.0 release. 
There's a fix in the other thread and a v0.3 of the Focus theme is about to be published. When the next version of Octodash comes out I'll reimplement that little icon.