UnchartedBull/OctoDash #1823: sudo missing from example update script in wiki

stephenobrochta, 2021-05-19:
Thanks for maintaining this software.

https://github.com/UnchartedBull/OctoDash/wiki/Update#in-app-update

The code for the bash script is missing `sudo`
instead of `dpkg -i /tmp/octodash.deb` ...
`sudo dpkg -i /tmp/octodash.deb`

UnchartedBull, 2021-05-20:
That script shouldn't include `sudo` since it will be called from OctoDash itself (so there is no way to enter a password). As mentioned in the text below that passwordless sudo needs to be enabled for this to work.

closed 2021-05-20 by UnchartedBull

stephenobrochta, 2021-05-21:
I must be confused about something in this process. With passwordless sudo enabled, the update script hangs as described in issue [1211](https://github.com/UnchartedBull/OctoDash/issues/1211). I assumed this was because `dpkg` must be run as root and calling the script from octodash is executing the script without superuser privilege. After adding `sudo`, from the command line I can execute the script without a password.

UnchartedBull, 2021-05-25:
It is definitely not needed for OctoPi and Raspbian (since many people ran the update just fine without `sudo` in that line). What OS are you running?

stephenobrochta, 2021-05-26:
I realize it was my problem.

My syntax was wrong for passwordless `sudo`. The only reason the update script worked without password prompt is the password timeout had not occurred. Thank you for responding.