UnchartedBull/OctoDash #1827: Display Layer Progress no longer working after update

kats-greg, 2021-05-19:
After doing the update to 2.2.0  on my 7 instances of OctoDash, the Layers displays "Layer 0 of ---"

kats-greg, 2021-05-20:
> Do all 7 instances show the same behavior? Is the LayerProgress
> correctly shown in the OctoPrint WebUI? Is the Fanspeed correct in
> OctoDash?

All 7 are acting the same.  The layer is not displaying on Octodash and 
the fan speed is not displaying there either.

But the layers do display in octoprint.

I've attached two pics, one the web interface, and the other the 
octodash screen.

Thanks

PabloAVazquez, 2021-05-23:
I've updated this week to the latest version and only now am I printing something. First thing I noticed was when running the filament change routine... chose the new filament, went on to the heating up nozzle screen and it got stuck there showing 20 / 180, it didn't automatically retract the filament, when I noticed it wasn't doing it, my first thought was that the heater cartridge had failed, but octoprint in the computer showed the proper temperature, and it had already settled at 180 degrees, upon pressing "skip" on the TFT screen it retracted and could finish the filament change. Next surprise was when I set it to print! I chose the file, smooth Lego block animations look great! thumbnail, everything perfect... but once I pressed on the printer icon... it just reversed to the home screen!! reading 20 degrees for hot end and bed, 0 for fan,  operational status, but as you can see in the attached picture, the printer is working happily, octoprint reports all temperatures properly and reports operational status to be printing.

[image]

[image]

I'm running OctoDash 2.2.0
OctoPrint: 1.6.1
DisplayLayerProgress Plugin: 1.26.0 
Dashboard Plugin: 1.18.2 
Filament Manager: 1.7.2 
OctoDash Companion: 0.0.4 
OctoKlipper: 0.3.8.2 

Thanks!!!

Pablo

PabloAVazquez, 2021-05-23:
Interestingly, once the print job finished, I issued the "restart octodash" command and it is now displaying updated values!

UnchartedBull, 2021-05-25:
@kats-greg did you disable the websocket option in the DLP options by any chance? Or did you already resolve this?

UnchartedBull, 2021-05-25:
@PabloAVazquez please try using `localhost` if you encounter this issue again. Could be due to a flaky internet connection :)

kats-greg, 2021-05-25:
I did not disable it on purpose.

And no the issue is not resolved.

I am not seeing an option in OctoPrint for Websockets in DLP setting.  
Where would I find that?

Thanks

UnchartedBull, 2021-05-25:
add the very bottom :)

[image]

kats-greg, 2021-05-25:
> at the very bottom :)
>
> Screenshot 2021-05-25 at 19 36 28
> <https://user-images.githubusercontent.com/19818061/119543037-81780980-bd90-11eb-91e7-fdaefa6448c3.png>

That worked!

Thanks

PabloAVazquez, 2021-05-25:
@UnchartedBull yes, i used to have this problem of a flaky connection long ago, so i set it to localhost and port 5000 about 4 months ago.

Yesterday it also failed to update so this time, just as a test, I restarted octoprint, not octodash, and after showing connection problems while octoprint restarted, it established the connection automatically again and worked nicely! very odd. Tried it again right now and it's still working.

thanks @UnchartedBull !!!

UnchartedBull, 2021-05-25:
Ok it seems like everything is working now for everyone - that's great! Thanks for all the feedback.

Closing this here, if there are still errors feel free to comment here / open a new issue :)

closed 2021-05-25 by UnchartedBull