UnchartedBull/OctoDash #1828: Suddenly getting "initializing error"

6R0M7, 2021-05-20:
Hello.  I've been using Octodash for a few months now, and until today without any issues whatsoever.  It's awesome!  (I actually [just posted this](https://www.reddit.com/r/octoprint/comments/ndg1ze/octodash_running_in_custom_enclosure_and_wiring/) to Reddit advocating for more people to use Octodash.)

When I powered on my Pi/screen today, it went through the boot sequence then got stuck on a screen that says: INITIALIZING... Initializing is taking longer than usual.  Please make sure that Octoprint is running and that CORS is enabled for the API.

Octoprint is running, and I'm printing without issue from a web interface.  I don't know what is meant by CORS.  Rebooting has made no difference.

I'm not sure what version of Octodash I'm running, but it's presumably the latest (at least as of about 90 days ago).  Hardware is a Raspberry Pi4 and a BigTreeTech 5" screen.  Octoprint version is 1.6.0.

Any suggests on how to address/remedy this would be greatly appreciated!  Thanks.

jneilliii, 2021-05-20:
CORS is a setting inside OctoPrint's settings inside the API section at the top. Check the box to enable and restart OctoPrint and you should get past that message.

UnchartedBull, 2021-05-20:
Please enable CORS as mentioned by @jneilliii. If this doesn't solve your issue please reopen this issue and you'll receive more help. Thanks for spreading the word for OctoDash :)

closed 2021-05-20 by UnchartedBull

6R0M7, 2021-05-20:
> CORS is a setting inside OctoPrint's settings inside the API section at the top. Check the box to enable and restart OctoPrint and you should get past that message.

This worked.  Thanks for the quick response.  I never enabled that when installing Octodash, but never had an issue until yesterday.

smooreace, 2021-11-17:
Sorry to reopen an old wound, but I'm having a similar issue. The difference is that CORS is enabled. I tried disabling and re-enabling with restarts in-between with no luck. This seems to be the only place on the entire internet talking about this issue.

UnchartedBull, 2021-11-18:
@smooreace what version of OctoDash are you running? Please also make sure that you have websockets enabled (should be enabled by default though).

smooreace, 2021-11-20:
Stupid mistake. Somewhere along the line of moving things the USB cable was pulled from the printer. So it all boiled do to not being able to communicate with the printer. As soon as it was plugged in it worked fine.

Thanks!

burnerbmx, 2022-01-01:
I'm having issues with this exact issue if I unplug my screen everything boots and it connects to the printer but it just keeps happening it used to work great I did run a few commands trying to get it to run a 3" screen but not sure [image]

burnerbmx, 2022-01-01:
[image]

jneilliii, 2022-01-01:
Is it running on a remote OctoPrint or on the same pi that OctoPrint is installed on? If it's the same, you might want to change the URL to `http://localhost/` instead of the name `http://ginge.local`.

burnerbmx, 2022-01-01:
[image] 

It's like too me if I have my screen plugged in it stops my pi from connecting to the wifi or something going wrong in octodash as if the screen isn't on octodash won't boot up properly is there a command I can use to try upgrade to fix any problems ?

smooreace, 2022-01-01:
That initialization issue for me is 100% of the time related to the USB
cable connection. I fought with it for a week! Replaced the cheap POS cable
with something quality (not some cheap garbage from Amazon) and Ive never
had the problem again... unless I unplug it!