UnchartedBull/OctoDash #1844: Octodash nach Update nicht Dauerloop

cipi-sys, 2021-05-20:
**What doesn't work?**
Nach dem Update bleibt der Boot Vorgang hängen und selbst nach Stunden startet octodash nicht. Aber die Ladepunkte bewegen sich. 

**What did you already try?**
A clear and concise description of what you tried to make OctoDash work.
Mehrfach Neugestartet und vom Strom
Genommen. 

**General Information:**
 - Hardware [e.g. Raspberry Pi, if you have layout issues please also include your screen resolution] 
 Pi 4 4gb
 - OS Info [e.g. Raspbian Buster Lite, please also indicate if you used OctoPi]
 OctoPi
 - OctoDash Version [e.g. v1.0.0]
 - OctoPrint Version [e.g. v1.0.0]

**Additional context**
Add any other context or screenshots about the feature request here.

UnchartedBull, 2021-05-20:
Schalte mal bitte in den OctoPrint API Einstellung "CORS' ein und starte danach einmal neu, danach sollte alles funktionieren. Siehe auch hier: https://github.com/UnchartedBull/OctoDash/issues/1818.

Und GitHub issues am besten in Englisch machen, da dir so mehr Leute helfen koennen :)

cipi-sys, 2021-05-20:
Wow 🤩 das ging ja super schnell - Next Time in englisch Thank you 😍

closed 2021-05-20 by UnchartedBull