UnchartedBull/OctoDash #1846: Custom action for controlling Enclosure plugin outputs is missing

haplm, 2021-05-22:
**Describe the bug**
As discussed on Discord, the problem is that setOutput() method for the Enclosure plugin has no associated custom action, hence it can't be used from the UI.

Related ticket is #1123.
 
**Expected behavior**
I could set one of the Enclosure plugin outputs using a custom action.

ab-tools, 2021-05-23:
While adding output support for the enclosure plug-in please also add support for PWM outputs of the enclosure plug-in as well (not just the regular outputs).

I've already created a pull request to add back-end support for this to the `EnclosureService` here:
https://github.com/UnchartedBull/OctoDash/pull/1848

UnchartedBull, 2021-07-28:
https://github.com/UnchartedBull/OctoDash/wiki/Custom-Actions#predefined-actions wiki has been updated as well. Will be part of next release. If you can't wait feel free to build from main once the linked PR has been merged :)

PR #2094 (2021-07-28): add output actions
Fixes #1846
closed 2021-07-28 by UnchartedBull