UnchartedBull/OctoDash #1849: RE: On Screen Keyboad

sonjamichelle, 2021-05-23:
Regarding an On Screen Keyboard.

During setup Having multiple printers OctoDash uses octopi.local for ALL found printers. Obviously this isn't optimal, so you have to use the enter manually option. Which is fine, except when having to hunt down a spare keyboard or scavenge one off another PC temporarily.

You mention in issue #207  about an on screen keyboard

> You can either copy the sample.config.json from this repo to ‚~/.config/octodash/config.json‘ and adjust it to your needs or >use xdotool (more info here: https://github.com/UnchartedBull/OctoDash/wiki/Installation) to emulate a keyboard.

HOWEVER you give no details. WHERE do we find this sample.config.json in the repo???

The link you point us to has NO info regarding the on screen keyboard. OR at least none my aging mind and eyes can find.

You DO have an on screen keyboard branch, but again, finding details is a little difficult.

I did see in another issue that on screen keyboard was not yet supported and to use another work around?

Is it still unsupported at this time? IS there a way to enable it?

jneilliii, 2021-05-23:
There still isn't an onscreen keyboard that I've seen. I agree having a default config.json in the repo would be helpful. My octodash companion plugin allows for direct upload of it within the octoprint ui.

jneilliii, 2021-05-23:
I just extracted this from source on my phone, so might want to validate. 

https://gist.github.com/jneilliii/392dbc243e688e65c0eef56125c17d68

UnchartedBull, 2021-05-25:
The example config is linked here: https://github.com/UnchartedBull/OctoDash/wiki/Setup-&-Settings#creating-the-config-manually (here is a direct link to the example config: https://github.com/UnchartedBull/OctoDash/blob/main/src/app/config/config.default.ts). It isn't ideal to have multiple printers advertising the same address, so it might be worth to change those (OctoDash will work fine as long as you don't use `octopi.local`)

The onscreen keyboard is currently just a development branch, so it isn't available in any of the released builds (and isn't working properly either yet). So atm there is no way to open the onscreen keyboard. For further updates on that please see #762.

closed 2021-05-25 by UnchartedBull