UnchartedBull/OctoDash #1855: After Update to 2.2.0  Fan-speed not displayed

Sabifabi, 2021-05-24:
Command send via octoprint or printer or otodash fan starts turning on but always 0% are displayed. Works fine before updating

komandrik, 2021-05-24:
I confirm. I am getting the same error.

jneilliii, 2021-05-24:
This is because DisplayLayerProgress plugin is either not installed, or doesn't have the websocket option enabled.

komandrik, 2021-05-24:
> This is because DisplayLayerProgress plugin is either not installed, or doesn't have the websocket option enabled.

If I turn off DisplayLayerProgress then there is a dash on the screen instead of the fan speed percentage. When DisplayLayerProgress is on, the screen is always 0. It doesn't matter if the fan is on or not.

Sabifabi, 2021-05-24:
> This is because DisplayLayerProgress plugin is either not installed, or doesn't have the websocket option enabled.

Enable the websocket option worked fine. Thanks for your fast support

closed 2021-05-24 by Sabifabi

UnchartedBull, 2021-05-25:
Check #1827.