UnchartedBull/OctoDash #1856: Upper line menu icon not on the screen on Ver2.2

kefkbw, 2021-05-25:
**Describe the bug**
I have tried to upgrade to V2.2, but I got error message.
I have installed new Octopi and Octodash step by step.
I think it was installed successfully, but I can't get the upper line menu icons.(File, Filament, Control)
Lower line icons displayed and working good. I don't know what I missing while I install V2.2.
I don't have touch screen, but I didn't have any problem to use old version of Octodash. I have used wired mouse.
My screen is recycled from old laptop with hdmi controller. I have tried 4times from format sd card to final installation.

[image]

UnchartedBull, 2021-05-25:
Confirmed this. It should return to "no job running ...". If you want to see the icons (File, Filament, Control) you need to enable the Touchscreen option.

kefkbw, 2021-05-25:
I have used the old version with no problem.
The new version should turn on the touch screen function?
Any other way to use it?

kefkbw, 2021-05-25:
How Can I turn on the touch screen function without reinstalling it again?
Is there any way to turn it on? I can't see the control icon either.

UnchartedBull, 2021-05-25:
https://github.com/UnchartedBull/OctoDash/wiki/Setup-&-Settings#manually-adjusting-the-settings set touchscreen to `true`.

PR #1861 (2021-05-25): show no job running after print again (no touch)
Fixes #1856.
closed 2021-05-25 by UnchartedBull