UnchartedBull/OctoDash #1877: Black border with 1024x600 screen resolution

mapotter99, 2021-05-28:
**What doesn't work?**
There is a black border around OctoDash.  The screen resolution is 1024x600.

**What did you already try?**
Nothing

**General Information:**
 - Hardware Raspberry Pi, 1024x600 screen
 - OS Info OctoPi v0.18.0
 - OctoDash Version v2.2.0
 - OctoPrint Version v1.6.1
**Additional context**

mapotter99, 2021-05-29:
Sorry, I'm new to Raspberry Pi and ALL of this.  I figured out how to customize config.txt for my screen, either directly from the card in the boot folder, or using SSH at "sudo nano /boot/config.txt".  I added the following lines under the [all] subheader...

hdmi_cvt=1024 600 60 3
hdmi_group=2
hdmi_mode=87
hdmi_drive=2
hdmi_force_mode=1

I also uncommented (removed the # from the beginning) hdmi_force_hotplug=1.

Everything else I left at defaults.

Now to figure out why the system is so unstable.  I'm wondering if the "sudo apt update/upgrade" as recommended the Chris's Basement youtube video may have messed things up, so I'm trying again without it.  Other possibilities include the screen's power supply not providing enough power for everything, or the touch interface gumming up the works,

closed 2021-05-29 by mapotter99

mapotter99, 2021-05-29:
The problem was indeed "sudo apt update/upgrade".  AVOID THESE COMMANDS!  Leave the OctoPi installation alone, aside from raspi-config and the OctoPi software updater!

Daghis, 2022-10-30:
I keep running into this problem because I keep forgetting how I fix it.

In my case, uncommenting this line in `/boot/config.txt` addresses it for my 1024x600 screen:
```
# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
disable_overscan=1
```