UnchartedBull/OctoDash #1879: Plugin like OctoRelay e.o.

dumbozo, 2021-05-28:
Thank you for the effort you put into OctoDash.
I do however find it very limited for my use.
Most features are available on the printer itself and can be used through the display while the enhanced features of Octoprint provided by plugins are not available.
The main issue for me is that my printer, light and enclosure fans are controlled by relais that are plugged into the GPIO of the Raspi and these are controlled by several plugins. The main one is OctoRelay (very simple).
I know you have the plugin for PSU controll but that one is not useable after the latest upgrades.
Now when my printer is off (time out or after a finished print) I cannot use Octodash to turn it on. I have to go to my computer use Octorelay and go back to octodash to do what? I could have easily start the print with Octoprint.
And all the features of the printhead and bed are easily done through the printers UI.
So for me Octodash is limited in use.
I hope you can add some plugins as OctoRelay soon.
Thanx in advance.
grt,
L3x

jneilliii, 2021-05-28:
Looking at the OctoRelay plugin it does expose an API, so it would be technically possible to do this, but would take someone's effort to do the programming on the OctoDash side to achieve this and with a install base of ~1000 it seems maybe popular enough to consider.

https://github.com/borisbu/OctoRelay/blob/9d9d85ca041597e8a5cec5a98b53b4bbd100400f/octoprint_octorelay/__init__.py#L183-L216

dumbozo, 2021-05-30:
It would be nice if someine could programm this.
To be able to controll the relays from Octodash would realy improve functionality for OctoRelay users

devjulie, 2021-08-28:
This is something I’d love to see, I’m just building a Prusa Box enclosure for my MK3 and really want to control the fan speed and LED’s from octodash. I hope this gets implemented soon otherwise I’ll need to keep using octopod on my iPhone to control the enclosure which isn’t always convenient.

UnchartedBull, 2021-09-08:
The next release of OctoDash will enable outputs for the enclosure plugin. So until OctoRelay is integrated it might be worthwhile checking out the other plugin. Maybe it'll also work.